dfki-ric/better_launch #11: Parsing path-like strings to better launch when launching from ROS2

jessemorris, 2025-09-19:
Hi,

Thanks for this great package - i've just been playing around with it today and came across a strange (I think!) issue when trying to call a better_launch launch file from ROS2

NOTE: I understand this is NOT best practice but my current projects involve a lot of integration with existing ROS2 launch files and I don't (at this point) want to change all other projects.

I want to parse an argument "path" that represents an absolute file path to my better_launch launch file

`
from better_launch import BetterLaunch, launch_this

@launch_this
def first_steps(path=None):
    bl = BetterLaunch()

    bl.node(
        "examples_rclpy_minimal_publisher",
        "publisher_local_function",
        "my_talker",
        )
`

which I call as:

`
ros2 launch test_pkg bl_test_path.launch.py path:=/path/to/thing
`

This throws the error:

`
[NOTE] Launch file bl_test_path.launch.py got included from ROS2
[ERROR] [launch]: Caught exception in launch (see debug for traceback): invalid syntax (<unknown>, line 1)
`

Calling the same file using the bl system works fine and correctly parses any parameter (the ROS2 launch system is indeed broken).
Getting rid of the leading/trailing "/" allows the launch system to work but this is not ideal as the value of "path" no longer represents an absolute file path.

Any suggestions as to what is happening here? Am I missing something obvious?

I also note that if I do not include a default value for path, launching the launch file from ros2 as before always causes an error
`
NOTE] Launch file bl_test_path.launch.py got included from ROS2
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - TypeError: 'type' object is not iterable
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown
`
regardless whether the argument path is provided as part of the cmdline args (ie path:=<>).

Please let me know of any potential fixes or what you may think the issue is; if you point me in the right direction, I am happy to try and fix with a PR.

Cheers

NOTE: I am running ros2 humble.

jessemorris, 2025-09-21:
@ndahn 
This seems to have fixed the issue.

However, I have additionally identified another issue when launching `bl` files with ros2: after launching the launch file, the ros launch program finishes (almost) immediately and before the node (or I guess any process, I have only tested with running nodes) finishes. Vitally, this leaves the running process hanging and I have to kill it with `ps -9 <process>`.

This happens even when running the example:
` ros2 launch better_launch 08_better_turtlesim.launch.py`

From some investigation, what I believe is happening when a launch file purely written in `bl` is called from ros, there are no long-running ros_launch.Actions to monitor for the launch process to manage and therefore the launch context finishes executing as soon as the call to `wrapper.generate_launch_description` finishes running (or sometime immediately thereafter). 

I validated this by adding a short `TimerAction` in with the `LaunchDescription()` and the (ros2) launch process continues to be active until this timer is done, regardless of what processes are launched by `bl`. This is obviously not ideal :)

My first thought solution was to create some kind of intermediate Action that monitors the state of any process launched by bl and only finish when those processes die or a shutdown command is received? I am not sure yet how the bl launch process stays alive until the launched processes are complete, as I guess the ideal behaviour is to mimic this from ros2 launch.

Please let me know if you observe the same behaviour; since this is a bigger issue, I am happy to open another git issue and continue there. Again, please let me know your thoughts on a solution, and I am happy to contribute.

Cheers,

ndahn, 2025-10-07:
Hi @jessemorris, and thank you for your patience! I have found a clean solution for the problem where the launch service would not stay alive using the somewhat obscure OpaqueCoroutine action. In my testing this seems to have solved the problem, but please let me know once you can confirm.

I will reply regarding the global remaps in a separate issue. Thanks for the suggestion!

closed 2025-11-03 by ndahn